winlinvip/SimpleDHT #2: DHT22 in the roadmap?

mbeacom, 2016-04-19:
@winlinvip Do you anticipate extending this to support the DHT22 model?

winlinvip, 2016-04-21:
@mbeacom No plan, I have no time to play, 😄 
BTW: Seems there are lots of libraries about the DHT22, why not use them?

mbeacom, 2016-04-21:
@winlinvip Honestly, I was just excited to see your DHT11 package, since I've been following you on GH as of lately and I'm in the middle of working on a personal, side project using the DHT11 and 22s. Haha

Regardless,  Thanks for everything @winlinvip !

winlinvip, 2016-04-22:
So you are using Arduino to Sample the temperature and humidity by DHT11/22?

mbeacom, 2016-04-22:
👍 Ongoing sampling with a raspberry pi and Arduino to monitor temperature/RH!

winlinvip, 2016-04-22:
It works~ I have done that.
BTW: It's not possible for raspberry-pi to directly drive the DHT11(maybe use C is ok, but python is not), for the DHT11 use Nms interval as the protocol, while the Arduino is good to do such things.

You can use Serial for Raspberrypi to communicate with Arduino. https://github.com/winlinvip/SimpleSerial
Use Arduino to sample the DHT11 and send the data to RaspeberryPi by usb serial port.

winlinvip, 2017-06-03:
It seems the protocol between DHT22 and DHT11 is not very different, I can support DHT22, but I haven't get a DHT22, can you test it for me?

winlinvip, 2017-06-03:
I have support DHT22 in https://github.com/winlinvip/SimpleDHT/commit/32bf0286397590c57ff4b3f428aa2c0cc1a496ef, however I didn't test it yet, please use https://github.com/winlinvip/SimpleDHT/blob/master/examples/DHT22Default/DHT22Default.ino to test it.
If it's ok, please close this issue. Thanks~

mbeacom, 2017-06-05:
@winlinvip I will test the patch sometime this evening and let you know!

winlinvip, 2017-06-10:
Great, what's the result?

equake, 2017-06-30:
I'm trying to use this lib with an DHT22 without success. I get error 101 / 102 while trying to fetch temperature.

I can make some tests if you want!

winlinvip, 2017-07-02:
@equake Please do some tests.

winlinvip, 2017-07-02:
[image]

I will refine the protocol according to http://akizukidenshi.com/download/ds/aosong/AM2302.pdf

closed 2017-07-08 by winlinvip

Rayzor53, 2017-11-16:
Doesn't work below freezing. Returns an absurd answer -2856.

winlinvip, 2017-11-27:
@Rayzor53 Dup to #14

commit 83303d21 (2017-07-08): Fix #2, support DHT22 with more accurate data.
commit 5432b375 (2017-07-08): Feature/dht22 (#10)

* Refine code according to new protocol

* Refine the readme

* Fix #2, support DHT22 with more accurate data.

* refine code

* Refine example