winlinvip/SimpleDHT #4: Suggestion

gvacondio, 2017-06-02:
Hello,
I suggest to invert lines 66 end 67 In SampleDHT.cpp code. This is because, after the start signal, the sensor pull down the data line before the wait time of 30uS.
In attach you can see the effect: 

[image]

Inverting the two lines of code, after submitting the start signal, data line switch immediately in INPUT mode and the problem is avoided.

Regards
Gian Marcoi

winlinvip, 2017-06-03:
The specification says:
```
    // notify DHT11 to start: 
    //    1. PULL LOW 20ms.
    //    2. PULL HIGH 20-40us.
    //    3. SET TO INPUT.
```

It seems that we should keep the HIGH signal 20-40us, so we should pull high then delay(30us), right?

winlinvip, 2017-06-03:
https://cdn-shop.adafruit.com/datasheets/DHT11-chinese.pdf

[image]

winlinvip, 2017-06-03:
Oh, I understand your point, when we pull high, we should set to input mode then wait 20-40us. That make sense. Thanks~

closed 2017-06-03 by winlinvip

gvacondio, 2017-06-03:
Hello my friend,

yes and I'll try to better explain the issue:

In the following figure:
1 - signal pulled low from arduino: this is the start signal sent to dht11
2 - end of the start signal from arduino. From now the dht11 can send its response
3 - the dht11 pull down the line to indicate the start of data stream, but the line is still pulled up from arduino, still in output mode. The result is a signal of about 3,3 V with a contention between arduino and dht11
4 - arduino set the pin in input mode, the contention ends, The line is definitively pulled down by dht11
5 - the start signal from dht11 ends.
6 - from now we will have 32 low states of variable length, one for any byte to be sent. 

[image]

To solve the issue, we must switch lines 66 and 67 in the code. Doing so, we will obtain the following:

[image]

Arduino switch the pin in input mode at point #4. In this way we will not have line contention when dht11 will pull down the line with its start signal.

I hope to have been clearer than before.

Ciaoo
Gian Marco

winlinvip, 2017-06-04:
Yep, I'm really appreciate your detail description, it's very clear. 
Thanks~

nantosuelta, 2020-03-07:
My DHT22 won't read with an initial 1ms pull low time, I had to change it to 10 ms and it worked. 
SimpleDHT.cpp line 311