winlinvip/SimpleDHT #7: this is probably not an issue...

jackfruh, 2017-06-24:
I'm using Arduino IDE 1.8.3 on osx.

The static values aren't working 
I get an error 
  
    DHT11:26: error: 'SimpleDHT' has not been declared
       int err = SimpleDHT::ErrSuccess; 
                 ^
    DHT11:27: error: 'SimpleDHT' has not been declared

I'm assuming this is more an issue of my compiler than the code? If so any idea how to fix or what I'd search the net for? (I know I can hack it by substituting 0 for SimpleDHT::ErrSuccess, just wondering how to fix that the static class member's not being seen.)

jackfruh, 2017-06-25:
Digging in a little deeper, it looks like version 1.0.2 did not have the statics. 
When I looked at the zip file on my machine, it was the 1.0.2 version

1.0.2 is the highest version shown in the Arduino IDE's Library Manager.

Doing a bit of research it seems the library.properties file may need to be updated for this to show up in Arduino IDE Library Manager. I put in a pull request for this. 

(Note I am new to this, doing the best I can to find the cause/fix, but I may be way off)

bgiesbrecht, 2017-06-25:
Hi @jackfruh, you can download the latest version from GitHub and manually install it. I just did this myself yesterday. If you run into issues or have any questions, shout. I like this newer version better due to the error codes.

Just in case you need the docs for this: [https://www.arduino.cc/en/Guide/Libraries#toc5](https://www.arduino.cc/en/Guide/Libraries#toc5).

jackfruhecolab, 2017-06-26:
Thanks bgiesbrecht! I did that and it got me a little closer! (getting some sensor checksum errors, ordered new sensors to see if that fixes the issue)

winlinvip, 2017-06-28:
Maybe I shouldn't use the static field of class, maybe macro is ok.

winlinvip, 2017-06-28:
I've fixed it, please test it.

winlinvip, 2017-06-28:
For 1.0.4, please test:
1. Whether it fixed the build failed bug?
1. Whether it works for DHT22?

closed 2017-07-02 by winlinvip