winlinvip/SimpleDHT #9: Does not work with Teensy

Rotario, 2017-06-28:
Hi,
I've got a DHT22, running the DHT22Default example.
I'm trying to get this library to work with the Teensy 3.6 and I'm getting a standard error:
err=101

This is using a cloned library of this git so latest version I assume. The board works with the standard ADAFRUIT library but I'd prefer to use this one as I can access just the simple bits which would speed up the data collection. I've read that the Teensy is much more accurate with its delayMicroseconds and the CPU I'm running clocks at 180MHz whereas an arduino would be around 18MHz, maybe I should just slow down the transfer?
Thanks,
Rowan

winlinvip, 2017-06-30:
The error code 101 is defined:
```
// Error to wait for start high signal.
#define SimpleDHTErrStartHigh 101
```

That because the start high signal is not arrived according to https://cdn-shop.adafruit.com/datasheets/DHT22.pdf

[image]

If Teensy is more accurate, I think it should work. I will do some research.

winlinvip, 2017-06-30:
I note that there's a comment:

```
// NOTE: For working with a faster chip, like an Arduino Due or Teensy, you
// might need to increase the threshold for cycle counts considered a 1 or 0.
// You can do this by passing a 3rd parameter for this threshold. It's a bit
// of fiddling to find the right value, but in general the faster the CPU the
// higher the value. The default for a 16mhz AVR is a value of 6. For an
// Arduino Due that runs at 84mhz a value of 30 works.
// Example to initialize DHT sensor for Arduino Due:

DHT dht(DHTPIN, DHTTYPE, 25);
```

I think it's because it use for loop to simulate the delayMicroseconds, as the Teensy is a faster chip so it need to change the threshold. But we use delayMicroseconds and now that Teensy is much more accurate, I think the delayMicroseconds is also accurate, why should change the `delay(10)` to `delay(15)` and (30) to (35)?

winlinvip, 2017-07-08:
I found it, it's because the data format is changed. Read `Table 5：AM2302 Communication format specifier` of http://akizukidenshi.com/download/ds/aosong/AM2302.pdf

I will fix it in v1.0.5, please try it when released.

closed 2017-07-08 by winlinvip

winlinvip, 2017-07-08:
Read #2