winlinvip/SimpleDHT #13: First read always fail

nbenm, 2017-09-16:
Hi,

I'm using code from sample with DHT22:

`    if ((err = dht22.read2(pinDHT22, &temperature, &humidity, NULL)) != SimpleDHTErrSuccess) {  
      Serial.print("Read DHT22 failed, err="); Serial.println(err);delay(2000);  
      return;  
    }`  

It always fails the first time.
It's annoying because I use esp8266 in deepmode. So the code, as it's executed after reset, is always in a first execution.

I've changed my code to double the read as

`  if ((err = dht22.read2(pinDHT22, &temperature, &humidity, NULL)) != SimpleDHTErrSuccess) {  
    delay(1000);  
    if ((err = dht22.read2(pinDHT22, &temperature, &humidity, NULL)) != SimpleDHTErrSuccess) {  
      Serial.print("Read DHT22 failed, err="); Serial.println(err);delay(2000);  
      return;  
    }  
  }`  

It works but it's ugly and takes more time (more power is consumed).
What can I do else ?
I don't have a DHT11 to test.

Thanks in advance.

Noury

winlinvip, 2017-09-22:
Why does this happed?

nbenm, 2017-09-22:
I don't know why this happens.
That's why I need help.

nbenm, 2017-09-28:
I have lookup more deeply with a logical analyzer.
The issue happens at startup only, not after every reset.
The esp8266 starts too fast. It sends the start signal (low), but the dht22 doesn't reply. It's not ready to do this. A simple delay after power up solves the problem. One second is not sufficient, two is better.
The library is not involved at all.

closed 2017-09-28 by nbenm

winlinvip, 2017-10-16:
👍 Thanks for your work.