winlinvip/SimpleDHT #14: Negative temperatures

danielfaust, 2017-11-14:
Have you tested this library with negative temperatures?

With a DHT22 I'm getting values like -3276.3 vor values slightly below 0.

winlinvip, 2017-11-27:
Nop, I didn't test it. I'll try it in our refrigerator.

danielfaust, 2017-11-27:
Here is the code I'm using to work around this issue.

As I'm sending the data over BLE to an Android phone and the USB port of the ESP32 NodeMCU broke off (I'm using the library with that), making an on-device correction somewhat more laborious, i fixed it in the app which receives the data.

Before sending it to the phone I multiply with 100 and then cast to int, that makes the BLE decoding somewhat easier.

In the app I'm using the following to correct the issue:

```
int iTemp = characteristic.getIntValue(BluetoothGattCharacteristic.FORMAT_SINT32, 4);
if (iTemp < 0) {
    iTemp = -iTemp - 327700;
}
float fTemp = iTemp / 100.0;
```

sulicz, 2017-12-01:
Hi, solution si:
```
    if (ptemperature) {
        if ( (temperature & 0x8000 )) {
          *ptemperature = (float) (temperature & 0x7FFF)/ -10.0;
        } else {
          *ptemperature = (float) temperature / 10.0;
        }
    }
```
in function SimpleDHT22::read2  and I changed all datatype short to uint16_t but datatype is may not a reason of problem.

winlinvip, 2017-12-11:
Could u please file a PullRequest?

danielfaust, 2017-12-11:
I will look into it

danielfaust, 2017-12-15:
BTW, using a compressed gas duster (holding it upside down) can give you temps below the freezing point very quickly, then you can heat it up again with the fingers.

winlinvip, 2018-01-02:
Merged

winlinvip, 2018-01-02:
Please fix it for DHT11.

danielfaust, 2018-01-02:
I don't have a DHT11 to test with. But since the DHT11 only works with temperatures equal or above 0 °C, I think that sign flag never gets set. I preferred not to touch that code.

> [Good for 0-50°C temperature readings ±2°C accuracy](https://www.adafruit.com/product/386)

mapmelad, 2018-01-15:
Unfortunately still not displayed correctly :/ 

https://puu.sh/z23v3/7b8f96b37f.jpg

danielfaust, 2018-01-15:
@mapmelad on what device are you running it? For me this works on an ESP32, maybe this is device dependent? When you use a previous commit, what does it show?

mapmelad, 2018-01-15:
First, I obtained the values of the similarity -3277.0°C. 
I used the fix:
`if (ptemperature) {
        if ( (temperature & 0x8000 )) {
          *ptemperature = (float) (temperature & 0x7FFF)/ -10.0;
        } else {
          *ptemperature = (float) temperature / 10.0;
        }
    }`
Then, after the fix I started getting the values 255, 254 and so on
If I delete in the code the minus sign `*ptemperature = (float) (temperature & 0x7FFF)/ 10.0;` , it displays the temperature correctly, but there is no minus sign in the output.

danielfaust, 2018-01-15:
Maybe this is due to the ArduinoBT -- which you're apparently using, judging from the screenshot -- being an 8 bit mcu while the ESP32 has a 32 bit cpu.

winlinvip, 2018-01-17:
Should we reopen this issue? @danielfaust

winlinvip, 2018-08-18:
Please test 1.0.9

closed 2018-08-18 by winlinvip