winlinvip/SimpleDHT #15: Reading errors

tomiisp, 2017-12-14:
DHT22 with ESP8266 producing in my case a lot of repeating read errors (sample rate 5sec), error rate was about 9:1 (error:success). Little looking in code and here seems was that issue:

`int SimpleDHT::confirm(int pin, int us, byte level) {
...  
      delayMicroseconds(10);
...
}`

changing delayMicroseconds(10); to delayMicroseconds(20); fix this problem, 0 errors after 2hours of testing

winlinvip, 2018-01-02:
It may fix the problem, but I think it's not the right answer. I think we must find out why delay(10us) caused the error, right?

vitamins, 2018-01-16:
I tried to apply this and it didn't help, the number of errors increased.
The adafruit library is working fine for me.
https://github.com/adafruit/DHT-sensor-library

winlinvip, 2018-01-17:
I think maybe the confirm function, which split us to pieces of 10us, causes deviation.

glewellyn, 2018-01-18:
Again, no root cause, but increasing the delayMicroseconds to 20 did solve my problem as well.

winlinvip, 2018-08-18:
Please test 1.0.9

closed 2018-08-18 by winlinvip