winlinvip/SimpleDHT #18: Running on Mega2560 cause

nailgilaziev, 2018-02-09:
HI! Thanks for this great library! 

I want to report a strange behavior on mega2560. 
Same dht22, same pullUP 10K resistor, same wires, same Pin on board. 

**attempt №1** , I connect it to my Arduino Nano and run example sketch:

```
=================================
Sample DHT22 with error count
//...iterate 100 times
24.80 *C, 27.50 RH%, total: 100, err: 0, success rate: 100.00%
```

Success rate 100% after 100 iterates.

**attempt №2**,  I cennect it to my Arduino Mega Pro (Atmega2560) and run same sketch:

```
=================================
Sample DHT22 with error count
//...iterate 100 times
Read DHT22 failed, err=105
0.00 *C, 0.00 RH%, total: 100, err: 31, success rate: 69.00%
```

**attempt №3** 
Arduino Mega (Atmega 2560 / 16MHz)
18.10 *C, 91.80 RH%, total: 100, err: 29, success rate: 71.00%

nailgilaziev, 2018-03-05:
who can give advice in which direction should I look for a problem?

winlinvip, 2018-07-12:
👍 Please file a MergeRequest.

winlinvip, 2018-07-12:
I will test it on Arduino with DHT11 and DHT22.

delfuego, 2018-07-25:
@t-w Your fork works perfectly for me! Prior to it, with my MEGA2560, I was getting 95 read errors out of 100 (not an exaggeration); now, it's perfect.

@winlinvip, any chance of getting this merged in? Pretty please? If only because I experienced what my 10yo's reaction was when the DHT11 in our kit "wasn't working at all" — leading me to hunt and figure out what might be happening, and finding this.

winlinvip, 2018-08-14:
I need to test it on Arduino UNO R3.

winlinvip, 2018-08-18:
Please test 1.0.9

closed 2018-08-18 by winlinvip