winlinvip/SimpleDHT #20: Sample DHT22... Read DHT22 failed, err=101

vjkanth, 2018-02-25:
Kit: Wemos D1 Mini, Sensor DHT22

I've used the SimpleDHT code and have the required libraries. Compiled and uploaded OK but keep getting 
=================================
Sample DHT22...
Read DHT22 failed, err=101

I'm powering the sensor externally and connected it to D4 on my Wemos D1 Mini board. The code in the same to assign the pin is:

**int pinDHT22 = 2;**

I'm not sure what that 2 refers to. Is it D2, or GPIO2 or physical pin 2 (which is A0) can someone who's used this successfully please clarify?
Wemos Pinout is here: https://escapequotes.net/esp8266-wemos-d1-mini-pins-and-diagram/

Thanks in advance.

vjkanth, 2018-02-25:
Resolved! The PIN designation in the code above is the Arduino Pin reference. In Wemos it equated to D4. Works perfectly. Thanks for the code!
Temperature: 25.30 *C
Humidity: 32.00%
👍

closed 2018-02-25 by vjkanth

winlinvip, 2018-03-03:
👍