winlinvip/SimpleDHT #21: err 101

pouria-niknamvazifeh-mscst, 2018-04-30:
Hi! 
I'm using this code with Arduino Uno and a DHT11 but keep getting error code 101. I couldn't figure out a way to fix it.

t-w, 2018-07-05:
I had similar issues (https://github.com/t-w/SimpleDHT/issues/2, https://github.com/t-w/SimpleDHT/issues/3). If you haven't resolved this, you can try a version with my patch:
https://github.com/t-w/SimpleDHT/tree/mega2560_r3
or slightly different timing in one place on comm. init.:
https://github.com/t-w/SimpleDHT/tree/esp8266
(the second might work for more cases, since the error threshold is lower / the code is more tolerant).

If this works for you (and I will have such signals from others) I can open a merge request, to improve the mainline. For the moment the patched version is only very briefly tested.

winlinvip, 2018-08-18:
Please test 1.0.9

closed 2018-08-18 by winlinvip