winlinvip/SimpleDHT #24: undeclared functions when compiling for non-AVR boards

wergor, 2018-08-23:
when compiling for ESP8266 or ESP32 (others propably too) compilation fails with the following error:

> C:\Users\user\Documents\Arduino\libraries\SimpleDHT\SimpleDHT.cpp:72:27: error: no 'int SimpleDHT::getBitmask()' member function declared in class 'SimpleDHT'
> 
>  int SimpleDHT::getBitmask() {
> 
>                            ^
> 
> C:\Users\user\Documents\Arduino\libraries\SimpleDHT\SimpleDHT.cpp:76:24: error: no 'int SimpleDHT::getPort()' member function declared in class 'SimpleDHT'
> 
>  int SimpleDHT::getPort() {
> 
>                         ^

this is because the function definitions of SimpleDHT::getBitmask() and SimpleDHT::getPort() are not enclosed within a conditional compilation block, unlike their declarations in SimpleDHT.h

paulocotta, 2018-08-24:
I also had the same problem

int SimpleDHT::getBitmask() {
int SimpleDHT::getPort() {

closed 2018-08-25 by winlinvip

rolfbachmann, 2018-08-25:
Had the same problem too. Downgrad to 1.0.8 works for me.

winlinvip, 2018-10-14:
@rolfbachmann Please test 1.0.12

commit ee2b6281 (2018-08-25): Fix #24, Support none AVR boards