winlinvip/SimpleDHT #25: err 100

Beguineur, 2018-08-24:
I tried with arduino DTH11
rev 1.0.10
and got the following error

Sample DHT11...
Read DHT11 failed, err=100

I don't know what err 100 means

winlinvip, 2018-08-25:
It means: 
```
// Error to wait for start low signal.
#define SimpleDHTErrStartLow 100
```

I should refine the error code, combining the t.

Beguineur, 2018-08-25:
Thank you having improve !

I am not a specialist and I still don't understand
I tried rev 1.0.11
the result is as follows :
=================================
Sample DHT11...
Read DHT11 failed, err=4112
=================================
Sample DHT11...
Read DHT11 failed, err=3088
=================================
Sample DHT11...
Read DHT11 failed, err=4112
=================================
Sample DHT11...
Read DHT11 failed, err=3088
=================================
it still doesn't work ....
can you help ?
Philippe

winlinvip, 2018-10-14:
@Beguineur  Please test 1.0.12

Imperator26, 2019-06-30:
@winlinvip Could we open this one up again? I have the same issue and would like to know what err=4112 and err=3088 mean?

winlinvip, 2020-01-08:
@Imperator26 https://github.com/winlinvip/SimpleDHT/blob/master/SimpleDHT.h#L30

marcdraco, 2020-09-27:
@winlinvip 

I busted a DHT22 (no idea how ... but there you go) and I've seen both the 3088 and the 4112 errors.

Reading the code, xx12 makes sense but xx88? I can't see any of your defines where that's set. I'm curious as although the sensors unquestionably misbehaving, it does give some sort of reading (it's just off by a few degrees).

I use a switch... case ladder to help me with the issue (failing to pull start low) but I'll be darned if I get it to 3088 again - but even if I did, you don't define that! Any thoughts? I'm not really good enough to debug someone else's code.

winlinvip, 2021-01-05:
新增了宏定义获取错误时间和错误码：

```
// Get the time duration from error.
#define SimpleDHTErrDuration(err) ((err&0xff00)>>8)
// Get the error code defined bellow.
#define SimpleDHTErrCode(err) (err&0x00ff)
```

使用参考Example：

```
  if ((err = dht11.read(&temperature, &humidity, NULL)) != SimpleDHTErrSuccess) {
    Serial.print("Read DHT11 failed, err="); Serial.print(SimpleDHTErrCode(err));
    Serial.print(","); Serial.println(SimpleDHTErrDuration(err); delay(1000);
    return;
  }
```

commit d9fdf066 (2018-08-25): Fix #25, Combine error and t into error code.
commit 4ee05cef (2021-01-05): Fix #25, Support SimpleDHTErrCode and SimpleDHTErrDuration
commit 226c21d3 (2021-01-05): Fix #25, Support SimpleDHTErrCode and SimpleDHTErrDuration
closed 2021-01-05 by winlinvip