winlinvip/SimpleDHT #32: Example code question.

macdroid53, 2019-01-06:
Hi,

I am using SimpleDHT and it works fine.

My question is about this snippet in the examples:

`
void loop() {
...
if ((err = dht22.read2(&temperature, &humidity, NULL)) != SimpleDHTErrSuccess) {
    Serial.print("Read DHT22 failed, err="); Serial.println(err);delay(2000);
    return;
...
  }`

Why is there a return?

Doesn't this exit loop()? (And wouldn't that return to the somewhere in the boot process of the device?)

Thanks,
Mac

ostaquet, 2019-02-15:
Hi,

The return statement allows to avoid the output of the results on the Serial. With the return, the loop() method starts again without going through the setup().

KR,
Olivier

macdroid53, 2019-02-16:
Does that mean code in loop() following the return does not get executed?

ostaquet, 2019-02-17:
Yes, it doesn’t

closed 2021-09-27 by winlinvip