winlinvip/SimpleDHT #33: Where are the error codes listed?

quoctran98, 2019-02-14:
This library works fine for me, except in the beginning and randomly distributed throughout I will receive 4 digit error codes. Where can I look these up? Thanks.

ostaquet, 2019-02-14:
You will find all error code in the header file `SimpleDHT.h`.

```
// High 8bits are time duration.
// Low 8bits are error code.
// For example, 0x0310 means t=0x03 and code=0x10,
// which is start low signal(0x10) error.
// @see https://github.com/winlinvip/SimpleDHT/issues/25
#define simpleDHTCombileError(t, err) ((t << 8) & 0xff00) | (err & 0x00ff)
```

```
// Success.
#define SimpleDHTErrSuccess 0
// Error to wait for start low signal.
#define SimpleDHTErrStartLow 0x10
// Error to wait for start high signal.
#define SimpleDHTErrStartHigh 0x11
// Error to wait for data start low signal.
#define SimpleDHTErrDataLow 0x12
// Error to wait for data read signal.
#define SimpleDHTErrDataRead 0x13
// Error to wait for data EOF signal.
#define SimpleDHTErrDataEOF 0x14
// Error to validate the checksum.
#define SimpleDHTErrDataChecksum 0x15
// Error when temperature and humidity are zero, it shouldn't happen.
#define SimpleDHTErrZeroSamples 0x16
// Error when pin is not initialized.
#define SimpleDHTErrNoPin 0x17
```

If you have the error code 4112, it means 0x1010 in hex. So, the time duration is 0x10 and the error in _Error to wait for start low signal_.

Hope it helps ;-)

quoctran98, 2019-02-18:
Thank you! I found the hex error codes but I didn't know that the time duration was in the front!

winlinvip, 2019-05-13:
#25

closed 2019-05-13 by winlinvip