winlinvip/SimpleDHT #35: Getting error `SimpleDHTErrStartLow` unnecessarily

timcharper, 2019-02-22:
In my DHT sensor, I'm consistently seeing a low time of 23-24us measured here:

https://github.com/winlinvip/SimpleDHT/blob/1.0.12/SimpleDHT.cpp#L220

I'm not sure if it's because the microcontroller I'm using has timing issues, or it's an odd DHT sensor, but simply reducing the threshold from 30 to 20 is the difference between the sensor failing continuously, and consistently reading successfully.

timcharper, 2019-02-28:
@winlinvip I experimented with this and found the issue; `pinMode` can take a variable amount of microseconds to complete. Further, per the spec, you're supposed to leave the pin HIGH for 25 microseconds, and _then_ read (and expect a high value from the sensor within 80 microseconds).

I rewrote the timing code to use a stop-watch style approach and it made things much more stable; allowing the pinMode change to be included in the 80 microseconds for which we're waiting for the high signal from the sensor has completely erased the non-determinism I was seeing and I'm getting significantly more stable readings.

https://github.com/timcharper/UniversalDHT/blob/master/UniversalDHT.cpp#L132

winlinvip, 2019-05-13:
👍