winlinvip/SimpleDHT #37: DHT22 does not work inside sketch

vmajor, 2019-04-28:
I am using Arduino UNO.

The DHT22default sketch compiles and runs well.

I am attempting to apply the same approach as in the sketch but display the results on a 1.8" TFT.

I am however unable to achieve this as the TFT enabled sketch fails to read from the DHT22 sensor and gives me error 4112 or 3088 on serial or on TFT (I got that to work)

The DHT22 deployment is basically identical to the sketch except I used #define instead of int to bind the pin. #define works in the DHT22default  sketch when I made the same change.

So, basically when I run the DHT22default sketch, the sensor works, when I run the (seemingly) identical sketch modified for TFT output, the sensor fails.

Here is the sketch. I also included the serial output to help me debug. It did not help.

`

// DHT
#include <SimpleDHT.h>

// include TFT and SPI libraries
#include <TFT.h>  
#include <SPI.h>

// pin definition for Arduino UNO
#define cs   10
#define dc   9
#define rst  8

// DHT section
#define pinDHT22 12
 // int pinDHT22 = 12;
  SimpleDHT22 dht22(pinDHT22);

// create an instance of the library
TFT TFTscreen = TFT(cs, dc, rst);

void setup() {
  Serial.begin(9600);
    
  //initialize the library
  TFTscreen.begin();

  // clear the screen with a black background
  TFTscreen.background(0, 0, 0);
  //set the text size
  TFTscreen.setTextSize(1);
  TFTscreen.stroke(255,0,255);
  TFTscreen.text("=================================", 6, 6);
  TFTscreen.text("Sample DHT22...", 6, 14);
  delay (1000);
  TFTscreen.background(0,0,0);
  
}

void loop() {
  TFTscreen.background(0,0,0);
  float temperature = 0;
  float humidity = 0;
  int err = SimpleDHTErrSuccess;
  if ((err = dht22.read2(&temperature, &humidity, NULL)) != SimpleDHTErrSuccess) {
    Serial.print("Read DHT22 failed, err="); Serial.println(err);
  }
  String  errorValue = String(err);
  char errorPrint[10];
  if ((err = dht22.read2(&temperature, &humidity, NULL)) != SimpleDHTErrSuccess) {
   errorValue.toCharArray(errorPrint, 10); TFTscreen.text("Read DHT22 failed, err=", 6, 6); TFTscreen.text(errorPrint, 6, 14); delay(2000); return;
  }
  
  TFTscreen.text ("Sensor Ok", 6, 6);
  delay (2000);
  TFTscreen.background(0,0,0);

  Serial.print("Sample OK: ");
  Serial.print((float)temperature); Serial.print(" *C, ");
  Serial.print((float)humidity); Serial.println(" RH%");
  
  // DHT22 sampling rate is 0.5HZ.
  delay(2500);
  
}`

TotalKnob, 2019-10-18:
How is your DHT22 wired up? Also, have you tried disconnecting and reconnecting power from the DHT22 while the code is running?

vmajor, 2019-11-29:
> How is your DHT22 wired up? Also, have you tried disconnecting and reconnecting power from the DHT22 while the code is running?

Sorry for not replying for months. The notification ended up in some random sorted box in Gmail. I found it by accident :(

I entirely forgot what I did, but I will get back to it and will resume with questions once I relearn what I tried to learn.

nantosuelta, 2020-03-07:
I have a similar issue.
My DHT22 won't read with an initial 1ms pull low time, I had to change it to 10 ms and it worked.
SimpleDHT.cpp line 311
The returned errors were 0xB10 and 0XC10 which led me to believe the DHT22 didn't "get" the start signal and the return is garbage...

Fix:
SimpleDHT.cpp line 311 replace with
delay(10); // 10 ms delay, actually anything >1ms && <20ms would do

winlinvip, 2021-09-27:
Please try again, I have reverted the 1.0.14

closed 2021-09-27 by winlinvip