winlinvip/SimpleDHT #41: Can we make setPin() public?

gregg-cbs, 2020-05-13:
Hey,

Thanks for the lib.
I am interfacing your lib with my own class so I can have all temperature logic in one place.

Doing this:
`SimpleDHT11 dht11(2);`

Stops me from being able to do this:
```
class Temperature{
    public:
      void setChannel(int _channel){
        channel = _channel;
        dht11.setPin(_channel);
      }
    ...
```

Is there a another way to instantiate this?

winlinvip, 2021-01-05:
You can create a new instance to set the pin:

```

class Temperature{
    public:
      void setChannel(int _channel){
        channel = _channel;
        dht11 = SimpleDHT11(_channel);
      }
```

winlinvip, 2021-01-05:
Or, you can pass the pin to read:

```
    virtual int read(byte* ptemperature, byte* phumidity, byte pdata[40]);
    virtual int read(int pin, byte* ptemperature, byte* phumidity, byte pdata[40]);
```

For exmaple:

```
SimpleDHT11 dht11;
class Temperature{
    public:
      void setChannel(int _channel){
        channel = _channel;
      }
      void read() {
        dht11.read(_channel, ......);
      }
```

winlinvip, 2021-01-05:
Done

commit a5add330 (2021-01-05): Fix #41, public the setPin
closed 2021-01-05 by winlinvip