winlinvip/SimpleDHT #45: New Feature suggestion 😀 Link to Online Simulator for SimpleDHT

Puneethkumarsn, 2020-08-30:
Hi There,

thank you for the working code. We liked it. We are working on the Arduino emulator where users can freely run and check the libraries without the actual hardware or the sensor. This will be helpful to make a quick test or a decision on the run.

Example: For **Adafruit** library we have here a permanent link: https://wokwi.com/arduino/libraries/DHT-sensor-library where the users can just try the library and get a good feel of the outcome even before they buy the hardware/rig up the hardware. 

Example 1: https://wokwi.com/arduino/libraries/DHT-sensor-library/DHT-Tester
Example 2: https://wokwi.com/arduino/libraries/DHT-sensor-library/DHT-unified-sensor

Also, we would like to ask for your suggestions and feedback. We will be happy to implement the useful features to make it more user friendly and helpful to the Arduino community 😍

I hope, this will be helpful for the users, using this library. We will be happy to share the link, create a Pull Request for the readMe file (for example) to provide the permanent simulated pages for the SimpleDHT library. 

We will be happy to hear from you. Thank you :)

winlinvip, 2021-01-05:
Thanks, I will try it.

Puneethkumarsn, 2021-01-08:
I will surely look into it as soon as possible and come back to you. 
thanks for reporting

winlinvip, 2021-01-11:
I test it again, it works!

[image]

But, the url is a random number, too complex to share to remember.

winlinvip, 2021-01-11:
But, the project url https://wokwi.com/arduino/projects/286863323599733260 is a random number, too complex to share to remember.