winlinvip/SimpleDHT #51: [Improvement] [DHT11] It is possible to get tenths of temperature

martin-eden, 2022-02-12:
For DHT11

While rewriting library for myself I've discovered that there are tenths of degrees Celsius transmitted in low nibble of second temperature byte. It is encoded in BCD, so hex values are in range 00..09.

martin-eden, 2022-02-12:
I'm using this "DHT11" sensor from eBay: https://www.ebay.com/itm/221914097856

Despite specified range 0..50 °C it can measure negative temperatures and have temperature granularity 0.1 °C. Temperature sign is stored in bit 7 of second temperature byte.

[image]

DaleSchultz, 2023-01-19:
are you getting more accuracy than DHT11.read**2**()  ?
(read2 takes floats)

martin-eden, 2024-09-17:
As you figured it out in another issues, 

read2() returns floats but temperature granularity is 1.0, not 0.1.
https://github.com/winlinvip/SimpleDHT/blob/f938823ea9e0eaaa2deee98171097582c953aa33/SimpleDHT.cpp#L180

That's how I've done it:
```C
  // 2nd temperature byte: tenths of temperature in low nibble in BCD.
  Temperature = (float) ((int16_t) Data[2] * 10 + (Data[3] & 0x0F)) / 10;
  // 2nd temperature byte: bit 7 is temperature sign.
  if (Data[3] & 0x80)
    Temperature = -Temperature;
```
That [lines] in my "scrapyard" unmaintained repo.

[lines]: https://github.com/martin-eden/Embedded_Scrapyard/blob/cf430f682db0aca049927c83c1ed2bb7fdf2bfda/libraries/me_DHT11/me_DHT11.cpp#L129-L133